weliem/bluez_inc #2: Functionality request related to Advertising

FcoFabra, 2022-08-30:
Thank you for this amazing library.
I am using it in a Raspberry Pi to connect using Android, and it works fine.

I also tried to connect from Windows using "Bluetooth LE Explorer" and I was unable to do that.
Moreover, in the advertising I include the device name. Android is able to find and parse the name, but the previous application not.

The next step was to use "BLE Scanner" in Android to analize the advertising, and I found that it includes the name of the device, but it does not includes the flags section (data type 0x01). I know that the standard makes this section optional, but it would be great to configure it in order to prepare a more "typical" advertisement.

Are you planning to implement this feature?
Is it possible at all?

Thank you for your support.
Best regards.

weliem, 2022-08-30:
Hi, no concrete plans to add this. As far as I am aware, using Flags can be done but that dinctionality is marked 'experimental'. 

Generally speaking I am staying away of experimental features. But perhaps.there is a way to support it without breaking stuff for people that have not enabled experimental features.....

FcoFabra, 2022-08-30:
Thank you.
I understand.

May be in the future.
Have a nice day.

closed 2022-08-30 by FcoFabra