weliem/bluez_inc #4: multiple same services

limitbit, 2022-11-01:
hi i want to create a hardware rev and software rev character to 'device info services', part of Bluetooth Base UUID (xxxxxxxx-0000-1000-8000-00805F9B34FB). i could advertise service with success, unfortunately there is multiple "Device Information" services on advertisement. I assume device already create a service  based on "Device Information UUID" with "PnP ID" character.

According to google this Pnp id;" This interface represents the Pn PID GATT Characteristic with uuid 2A50. It is referenced by the [DeviceInformation](https://docs.bosch-iot-suite.com/edge/edge-services/API/modules/com.prosyst.mbs.bluetooth.le.driver.api/com/prosyst/mbs/services/btle/gatt/service/DeviceInformation.html) service."

So the problem is, due to double "DEVICE INFO" services characteristic not found on central side. App found "Device Service" based on UUID unfortunately this service contains only PnP ID character.

I could not find a solution to add hardware rev and software rev char to single Device Information Service, and i need help. 
Thank you.

weliem, 2022-11-03:
hi, yes I also noticed that. Bluez is automatically doing that and it is not caused by this library. However, I haven't figured out a way to solve it...

limitbit, 2022-11-03:
ok, thanks for your answer. I am still digging.

abqjln, 2024-08-19:
What an awesome library, thank you! I am running into this issue as well as I build a Polar H10 emulator. Occasionally my DIS using bluez_inc works, but it too frequently fails. I've tracked it down to the doubly-linked GList in binc_service_get_characteristic. (it seems) bluez inserts a 2a50 as the first and last element in the list, so any search not for 2a50 fails.

I inserted a number of log_debug statements adding __func__ at the start--

FAILS (apparently using bluez auto-installed service/chars)
DEBUG [Device] binc_device_get_characteristic: Found service 0000180a-0000-1000-8000-00805f9b34fb
DEBUG [Service] binc_service_get_characteristic: Searching for characteristic <00002a26-0000-1000-8000-00805f9b34fb>
DEBUG [Service] binc_service_get_characteristic: Found characteristic 00002a50-0000-1000-8000-00805f9b34fb
DEBUG [Device] binc_device_read_char, service 0000180a-0000-1000-8000-00805f9b34fb, characteristic 00002a26-0000-1000-8000-00805f9b34fb, is null
DEBUG [gymlink.c] on_services_resolved: (pi3dev) could not read characteristic 00002a26-0000-1000-8000-00805f9b34fb`

WORKS (same code, different run, iterates through the list properly apparently using my services/chars)
DEBUG [Device] binc_device_get_characteristic: Found service 0000180a-0000-1000-8000-00805f9b34fb
DEBUG [Service] binc_service_get_characteristic: Searching for characteristic <00002a26-0000-1000-8000-00805f9b34fb>
DEBUG [Service] binc_service_get_characteristic: Found characteristic 00002a50-0000-1000-8000-00805f9b34fb
DEBUG [Service] binc_service_get_characteristic: Found characteristic 00002a27-0000-1000-8000-00805f9b34fb
DEBUG [Service] binc_service_get_characteristic: Found characteristic 00002a26-0000-1000-8000-00805f9b34fb
DEBUG [Device] binc_device_read_char: service 0000180a-0000-1000-8000-00805f9b34fb, characteristic 00002a26-0000-1000-8000-00805f9b34fb, supports = 1
DEBUG [Characteristic] binc_characteristic_read: Reading characteristic <00002a26-0000-1000-8000-00805f9b34fb>
DEBUG [gymlink.c] on_services_resolved: (pi3dev) Read characteristic 00002a26-0000-1000-8000-00805f9b34fb

I'll keep poking around. Is there a way to delete a characteristic (binc_service_delete_characteristic?) ? Alternately is there a way to remove the 0x180a service and reinstall at run time (binc_application_delete_service)? I'll keep poking around.

First time ever commenting--please correct any impertinence!

abqjln, 2024-08-20:
SOLVED: Asked and answered. I think.
Disable the auto-registering of DIS(0x180a) and PnP (0x2a50) under bluetoothctl: 5.66
[ https://github.com/bluez/bluez/issues/101](url)

Tested lightly on Raspberry Pi 3 (peripheral) and 4 (central) bookworm 32 bit

Two mods to /etc/bluetooth/main.conf
DeviceID=false
Cache=no (probably not necessary but convenient when developing)

sudo systemctl daemon-reload
sudo systemctl restart bluetooth
sudo systemctl status bluetooth
(or reboot)

weliem, 2024-08-20:
Ah great find. Will test this

weliem, 2024-08-25:
Works! Closing issue....

closed 2024-08-25 by weliem