weliem/bluez_inc #5: extern C for header files

bp1001, 2023-01-06:
Hi,

Thank you alot for your work.
Would it be possible to add the 

#ifdef __cplusplus
extern "C" {
#endif

#ifdef __cplusplus
}
#endif

around the public header files to make it useable with c++? 

Preferably only put them around the function prototypes, not the includes. New versions of Glib apparently include C++ headers, so that putting the extern "C" around the glib includes generates compile errors.

Regards

weliem, 2023-01-06:
Feel free to raise a PR! Then you can immediately check if it works as expected...

bp1001, 2023-01-12:
I have added the changes locally now, everything seems to be working fine. I tried to push the changes to a new branch, but I do not seem to have the access rights to do so.

weliem, 2023-01-14:
The way it works on Github is that you first 'fork' a repo. Then you make changes and raise a PR back to the original repo....

weliem, 2023-01-16:
Thanks for the PR, closing this issue

closed 2023-01-16 by weliem