weliem/bluez_inc #8: cannot register new advertisement

Sia1997, 2023-05-08:
Hi,
Thank you for your work on this library, it helps me a lot.

However, when I try to run peripheral so that I could establish a ble connection, I always got an error like "DEBUG [Adapter] failed to register advertisement (error 36: GDBus.Error:org.bluez.Error.NotPermitted: Maximum advertisements reached)" caused by binc_adapter_start_advertising.

Could you have a look over this?

Thanks!

weliem, 2023-05-08:
I think this happens because the advertisements are not being unregistered. Registering and unregistering happens in start/stop advertising.

So for every time you call `binc_adapter_start_advertising` , you need to call `binc_adapter_stop_advertising`.

weliem, 2023-05-13:
It looks like you are calling 'stop_advertising' when you didn't do a 'start_advertising'. Nonetheless, this error is 'innocent' and you should not see your original problem anymore, right?

Sia1997, 2023-05-15:
After tracing the error, it turned out to be the problem of hard-ware. Thanks for your help!

closed 2023-05-15 by Sia1997