weliem/bluez_inc #12: GDBUS error

tchen61, 2023-10-02:
example/central.c produced

2023-10-02 19:20:30:669 DEBUG [Adapter] failed to call adapter method (error 36: GDBus.Error:org.bluez.Error.InvalidArguments: Invalid arguments in method call)
2023-10-02 19:20:30:787 DEBUG [Main] discovery 'started' (/org/bluez/hci0)

(process:23652): GLib-CRITICAL **: 19:20:30.962: g_str_has_prefix: assertion 'str != NULL' failed
^C2023-10-02 19:20:33:587 ERROR [Main] received SIGINT

weliem, 2023-10-03:
The central.c application is working fine on my machine. So can you provide more information about when this error occurs? Did you change any code? Which adapter method is giving you the error?

You can also try to see if everything works when using the 'bluetoothctl' command.

tchen61, 2023-10-03:
i made a couple changes, otherwise the code would not compile: (agent.c and
application.c) to include errno.h
otherwise, no change

i am running ubuntu 18, bluetooth 5.53

weliem, 2023-10-03:
Right. Bluez 5.53 is super old and definitely not supported. The current version is 5.70. Try upgrading...

Not sure if it will work on Ubuntu 18 either. I am 22.04....but I'd first try to see if upgrading Bluez to a newer version solves your problem.

The examples compile file on my machine and no changes should be necessary

closed 2023-11-04 by weliem