weliem/bluez_inc #14: How to implement secure characteristic?

quangtv10, 2023-10-09:
Hello.
In my project, I'd like to create a GATT service with some characteristics.
And I want to make them secure (ask pairing to access them). Unsecure characteristics work fine.
Is it Possible ?
Is it possible to create secure characteristics with your framework?

Thank you

lizziemac, 2023-10-10:
I believe you'd want to make the characteristic part of a service that isn't advertised. For example, using a modified `peripheral.c` example:
```c
        // Setup advertisement
        GPtrArray *adv_service_uuids = g_ptr_array_new();
        g_ptr_array_add(adv_service_uuids, HTS_ADVERTISING_SERVICE_UUID); // Make a service specifically for advertising

        advertisement = binc_advertisement_create();
        binc_advertisement_set_local_name(advertisement, "BINC");
        binc_advertisement_set_services(advertisement, adv_service_uuids); // only add that service to the advertising services
        g_ptr_array_free(adv_service_uuids, TRUE);
        binc_adapter_start_advertising(default_adapter, advertisement);

        // Start application
        app = binc_create_application(default_adapter);
        binc_application_add_service(app, HTS_ADVERTISING_SERVICE_UUID);
        binc_application_add_service(app, HTS_SERVICE_UUID); // this is your real service, DON'T add it to the advertisement
        binc_application_add_characteristic(
                app,
                HTS_SERVICE_UUID,
                TEMPERATURE_CHAR_UUID,
                GATT_CHR_PROP_READ | GATT_CHR_PROP_INDICATE | GATT_CHR_PROP_WRITE);
```
or at least that's how I'm doing it, which results in one service being advertised until the device is paired, there may be more to it that Martijn might know.

weliem, 2023-10-12:
Unfortunately it is not (yet) possible to create characteristics that require encryption. I will put that on the backlog....

quangtv10, 2023-10-13:
Thanks @weliem , @lizziemac ,

I have tried as below and it's working

modified peripheral.c:

agent = binc_agent_create(default_adapter, "/org/bluez/BincAgent", DISPLAY_YES_NO);

modified application.c:
-------------- /binc/application.c --------------
index bf36272..90126e8 100644
@@ -697,7 +697,7 @@ static GList *permissions2Flags(const guint8 permissions)
     GList *list = NULL;
 
     if (permissions & GATT_CHR_PROP_READ) {
-        list = g_list_append(list, g_strdup("read"));
+        list = g_list_append(list, g_strdup("encrypt-read"));
     }
     if (permissions & GATT_CHR_PROP_WRITE_WITHOUT_RESP) {
         list = g_list_append(list, g_strdup("write-without-response"));
      
modified characteristic.c:
------------- binc/characteristic.c -------------
index dcb1e83..577f846 100644
@@ -521,7 +521,7 @@ static guint binc_characteristic_flags_to_int(GList *flags) {
             char *property = (char *) iterator->data;
             if (g_str_equal(property, "broadcast")) {
                 result += GATT_CHR_PROP_BROADCAST;
-            } else if (g_str_equal(property, "read")) {
+            } else if (g_str_equal(property, "encrypt-read")) {
                 result += GATT_CHR_PROP_READ;
             } else if (g_str_equal(property, "write-without-response")) {
                 result += GATT_CHR_PROP_WRITE_WITHOUT_RESP;

weliem, 2023-10-14:
I just committed a fix for this issue

weliem, 2023-10-22:
Closing issue

closed 2023-10-22 by weliem