weliem/bluez_inc #15: Can this library work on QCS404DK board?

IHNEL, 2023-10-19:
I run the peripheral code on the QCS404DK board () but couldn't register advertisement as attached log below, everything is working fine on Ubuntu 22.04. The QCS404DK board also runs Linux (see the output of `uname -a`:
>/ # uname -a
Linux nf-64 4.14.117-perf #1 SMP PREEMPT Mon Jan 4 17:55:28 CST 2021 aarch64 aarch64 aarch64 GNU/Linux

```
2021-01-04 09:43:30:812 DEBUG [Adapter] finding adapters
2021-01-04 09:43:30:816 DEBUG [Adapter] found 1 adapter
2021-01-04 09:43:30:816 DEBUG [Main] using default_adapter '/org/bluez/hci0'
2021-01-04 09:43:30:817 DEBUG [Main] Power status = 0
2021-01-04 09:43:30:819 DEBUG [Main] Power on adapter /org/bluez/hci0
2021-01-04 09:43:30:819 DEBUG [Main] Adapter name: hci0
2021-01-04 09:43:30:850 DEBUG [Application] successfully published application
2021-01-04 09:43:30:850 DEBUG [Application] successfully published local service 00001809-0000-1000-8000-00805f9b34fb
2021-01-04 09:43:30:850 DEBUG [Application] successfully published local characteristic 00002a1c-0000-1000-8000-00805f9b34fb
2021-01-04 09:43:30:851 DEBUG [Application] successfully published local descriptor 00002901-0000-1000-8000-00805f9b34fb
2021-01-04 09:43:30:851 DEBUG [Application] set value <68656c6c6f20746865726500> to <00002901-0000-1000-8000-00805f9b34fb>
2021-01-04 09:43:30:851 DEBUG [Adapter] failed to set adapter property (error 36: GDBus.Error:org.bluez.Error.Failed: Operation failed)
2021-01-04 09:43:30:851 DEBUG [Adapter] failed to register advertisement (error 19: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Unknown method 'RegisterAdvertisement' or interface 'org.bluez.LEAdvertisingManager1'.)
2021-01-04 09:43:30:857 DEBUG [Adapter] failed to register application (error 19: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Unknown method 'RegisterApplication' or interface 'org.bluez.GattManager1'.)
2021-01-04 09:43:32:192 DEBUG [Main] powered 'on' (/org/bluez/hci0)
```

weliem, 2023-10-21:
It looks like you are on an old version of Bluez. Try upgrading to the latest version.

IHNEL, 2023-10-23:
hi Weliem, which is the minimum Bluez version to make this library works?

weliem, 2023-10-23:
it depends a bit on which functions you use. Bluez is still changing with every version and adding new features. Try Bluez 5.65 or higher

closed 2024-01-26 by weliem