weliem/bluez_inc #16: Problems with notification

krimp, 2023-10-29:
I am trying to use this library to implement a Central for the Nordic Uart GATT on a RPi4B 64bit, Bullseye, Bluez 5.70. 
I am using the Central-example as the starting point.

The Central  is very simple: The RX has a notification only, and the TX a write:

`2023-10-29 13:13:28:562 DEBUG [Device] Characteristic{uuid='6e400003-b5a3-f393-e0a9-e50e24dcca9e', flags='[notify]', properties=16, service_uuid='6e400001-b5a3-f393-e0a9-e50e24dcca9e, mtu=247'}'

'2023-10-29 13:13:28:562 DEBUG [Device] Descriptor{uuid='00002902-0000-1000-8000-00805f9b34fb', flags='[]', properties=0, char_uuid='6e400003-b5a3-f393-e0a9-e50e24dcca9e'}'
2023-10-29 13:13:28:563 DEBUG [Device] Characteristic{uuid='6e400002-b5a3-f393-e0a9-e50e24dcca9e', flags='[write-without-response, write]', properties=12, service_uuid='6e400001-b5a3-f393-e0a9-e50e24dcca9e, mtu=247'}`

krimp, 2023-10-29:
Ahhh. Solved it:
Had to NOT convert the byteArray in the notification callback, due to **binary data**.
Had to access the received data by 
```c
byteArr->data
``` 
and

```c
byteArr->len
```

weliem, 2023-10-29:
Ok, good that you got it to work!

closed 2023-11-01 by weliem