weliem/bluez_inc #19: Set PIN for peripheral

lizziemac, 2024-01-19:
I didn't see anything in the code or documentation, but it's possible I might have missed something - does this library support setting a PIN for peripheral devices yet? If so, does it have to be 6 digits, or can it be 4?

weliem, 2024-01-20:
Hi, returning a 6 digit PIN code is supported. See the section in the Readme on Bonding.

lizziemac, 2024-01-20:
Thanks Martijn!

Oh yes, I see that it works the same way exactly for peripherals, I wrongly assumed the bonding section only applied to central devices.

I do have another question now that I'm a little further along: is there a way to set a custom passkey when using the `DISPLAY_ONLY` method on a peripheral?  If that's not currently implemented 

Thanks again for all of your help!
Liz

lizziemac, 2024-01-22:
Never mind, after some research I found that this is not a recommended practice anyways.

closed 2024-01-22 by lizziemac

lizziemac, 2024-09-17:
@GrubbyHalo This was the deciding conversation for me https://stackoverflow.com/questions/59280473/how-to-hard-code-ble-pairing-passkey-in-bluez-5-x-peripheral