weliem/bluez_inc #21: What are the hardware requirements of bluez_inc?

ludwit, 2024-02-14:
Hey weliem,

I was running your example BLE applications on my ubuntu 22.04.3 pc which has bluetooth 5.2 hardware support and a debian 11 beagle-bone-black with a bluetooth adapter supporting bluetooth 5.0.

Running the periperal executable on the bealge bone black I got this error message:
DEBUG [Adapter] failed to register application (error 36: GDBus.Error:org.bluez.Error.Failed: No valid service object found)

Whereas on the ubuntu pc the peripheral applicaton works fine. I can discover and connect to the device and retrieve the characteristc data using a generic BLE discover software with GUI (LightBlue). In this setup the central executable running on the beagle bone black isn't connection to the ubuntu pc's BLE service and does not retriev the "hello there".

So my question is wether you can name some hardware and/or software requirements for using bluz_inc, as the example works withthe ubuntu pc but not on the debian begle-bone-black?

Thanks in advance!

Appendix - all log for error message
debian@BeagleBone:~/bluez_inc/examples/peripheral$ ./peripheral
2024-02-14 10:55:42:978 DEBUG [Adapter] finding adapters
2024-02-14 10:55:42:988 DEBUG [Adapter] found 1 adapter
2024-02-14 10:55:42:989 DEBUG [Main] using default_adapter '/org/bluez/hci0'
2024-02-14 10:55:42:993 DEBUG [Application] successfully published application
2024-02-14 10:55:42:998 DEBUG [Application] successfully published local service 00001809-0000-1000-8000-00805f9b34fb
2024-02-14 10:55:43:001 DEBUG [Application] successfully published local characteristic 00002a1c-0000-1000-8000-00805f9b34fb
2024-02-14 10:55:43:010 DEBUG [Application] successfully published local descriptor 00002901-0000-1000-8000-00805f9b34fb
2024-02-14 10:55:43:016 DEBUG [Application] set value <68656c6c6f20746865726500> to <00002901-0000-1000-8000-00805f9b34fb>
2024-02-14 10:55:43:034 DEBUG [Application] adding /org/bluez/bincapplication/service0
2024-02-14 10:55:43:038 DEBUG [Application] adding /org/bluez/bincapplication/service0/char0
2024-02-14 10:55:43:040 DEBUG [Application] adding /org/bluez/bincapplication/service0/char0/desc0
2024-02-14 10:55:43:050 DEBUG [Adapter] started advertising (E8:48:B8:C8:40:00)
2024-02-14 10:55:43:063 DEBUG [Adapter] failed to register application (error 36: GDBus.Error:org.bluez.Error.Failed: No valid service object found)
2024-02-14 11:05:43:105 DEBUG [Application] freeing application /org/bluez/bincapplication
2024-02-14 11:05:43:110 DEBUG [Application] freeing service /org/bluez/bincapplication/service0
2024-02-14 11:05:43:111 DEBUG [Application] freeing characteristic /org/bluez/bincapplication/service0/char0
2024-02-14 11:05:43:111 DEBUG [Application] freeing descriptor /org/bluez/bincapplication/service0/char0/desc0

weliem, 2024-02-14:
Hi, I see it doesn't work on the Beagle. But it should. The only thing this library does is talk to the DBus...

Maybe it is a permission issue?

weliem, 2024-02-14:
Can you share some code? How are you starting the advertising?

ludwit, 2024-02-14:
I am using the code central and peripheral code from /examples.
Could it be the version of bluz? The beagle bone only had 5.55 While the ubuntu had 5.64.
YOu tested on raspberry pie i guess. What os did you use?

ludwit, 2024-02-14:
and
ebian@BeagleBone:~/bluez_inc$ tail /var/log/syslog
Feb 14 17:19:52 BeagleBone bluetoothd[1068]: sap-server: Operation not permitted (1)
Feb 14 17:19:53 BeagleBone systemd[1]: Started Load/Save RF Kill Switch Status.
Feb 14 17:19:53 BeagleBone systemd[1213]: Reached target Bluetooth.
Feb 14 17:19:53 BeagleBone systemd[1]: Reached target Bluetooth.
Feb 14 17:19:58 BeagleBone systemd[1]: systemd-rfkill.service: Succeeded.
Feb 14 17:20:43 BeagleBone dbus-daemon[610]: [system] Connection :1.22 (uid=0 pid=31852 comm="dbus-monitor --system destination='org.bluez' send") became a monitor.
Feb 14 17:20:47 BeagleBone bluetoothd[1068]: src/gatt-database.c:parse_chrc_flags() Invalid characteristic flag: encrypt-indicate
Feb 14 17:20:47 BeagleBone bluetoothd[1068]: src/gatt-database.c:chrc_create() Failed to parse characteristic properties
Feb 14 17:20:47 BeagleBone bluetoothd[1068]: src/gatt-database.c:client_ready_cb() No valid external GATT objects found
Feb 14 17:20:51 BeagleBone dbus-daemon[610]: [system] Monitoring connection :1.22 closed.

weliem, 2024-02-14:
Try removing the ENCRYPT_INDICATE flag. That was added in later versions of Bluez and in the log you see it crashes on it.....

ludwit, 2024-02-15:
I removed the GATT_CHR_PROP_ENCRYPT_INDICATE flag in the peripheral.c. Now the peripheral appliaction works.
Thank you for that!

Also I have a raspberry pi with bluez 5.66 up and running (the peripheral application works out of the box). Unfortunately although I the peripheral application works with beagel bone, ubuntu pc and raspberry pi the central application is not connecting or retrieving data on either of those machines although I can find the services via LightBlue. Do you have any idea why that could be? How was your setup when you tested it?

weliem, 2024-02-16:
The standard central app scans for a thermometer with a name prefix of "Taidoc". If you remove that the central will probably find the peripheral example.

ludwit, 2024-02-19:
Works now thank you a lot!

ludwit, 2024-02-19:
TLDR:
- If bluez is not outdated remove the ENCRYPT_INDICATE flag in binc_application_add_characteristic of the peripheral.c
- remove (NULL) or rename the pattern argument "TAIDOC" in binc_adapter_set_discovery_filter of central main.c

closed 2024-02-19 by ludwit