weliem/bluez_inc #22: `on_char_start_notify` is not being called when using `GATT_CHR_PROP_ENCRYPT_NOTIFY`.

KyleWatsonDev, 2024-02-15:
Hello Martijn,

I am trying to setup encrypted notifications using the `GATT_CHR_PROP_ENCRYPT_NOTIFY` characteristic property in my application, but it appears that when using the encrypted notify characteristic, notifications do not get enabled. When I just use `GATT_CHR_PROP_NOTIFY`, notifications are enabled and work totally fine. Within my code I am calling the following to setup notifications and register my callbacks for start and stop notify:

```cpp
Agent *agent = binc_agent_create(default_adapter, BLUEZ_AGENT_PATH, NO_INPUT_NO_OUTPUT);
// ...
binc_application_add_characteristic(app, SERVICE_UUID, CHAR_UUID,
         (GATT_CHR_PROP_NOTIFY | GATT_CHR_PROP_ENCRYPT_NOTIFY) |
         (GATT_CHR_PROP_READ | GATT_CHR_PROP_ENCRYPT_READ));
binc_application_set_char_start_notify_cb(app, &on_local_char_start_notify);
binc_application_set_char_stop_notify_cb(app, &on_local_char_stop_notify);
```

The encrypted read works fine and output is expected, but notifications are not being enabled. 

In the debug logs I would expect to see the following as when using just `GATT_CHR_PROP_NOTIFY`:

```
DEBUG [Application] read <UUID>
DEBUG [Application] start notify <UUID>
```

But instead with `GATT_CHR_PROP_ENCRYPT_NOTIFY` I just see:

```
DEBUG [Application] read <UUID>
```

I would also expect `binc_application_char_is_notifying(app, SERVICE_UUID, CHAR_UUID)` to return `TRUE` like it does when using unencrypted notifications, but it is returning `FALSE` when using encrypted notifications.

I have tested this on two separate devices, iOS and macos, and saw the same results.

Any help you can provide would be appreciated! Not sure if there is a step I am missing or if perhaps there is a bug here.

Thanks!
    Kyle

weliem, 2024-02-16:
Hi Kyle, to be honest I have no idea why it isnt working. When I am back from holidays I'll try to debug it.

KyleWatsonDev, 2024-03-18:
Hello Martijn,

I was just curious if you have been able to debug this issue at all or if you were able to replicate. I hope you had or are having a good holiday!