weliem/bluez_inc #23: Get CompanyID and Manufacturer Data on discovery callback

mamigot-salto, 2024-02-19:
Hello! First of all I'd like to thank you for the awesome work you've done with this bluez_inc library. Thank you so much!

Ok, here is the issue I'm struggling with and hopefully you can help me with: I'd like to parse as a byte array the manufacturer data that is included within the advertisement packet from a peripheral device, but the `binc_device_get_manufacturer_data` method returns a `GHashTable` which honestly I don't know how to process. What is the key that I should look for in that hash table so I can check the Company ID that is advertised? How can I parse the actual manufacture data that is included, is it a key or a value in the hash table? How is the data inserted in that hash table: as string or as byte array?

Thank you in advance

weliem, 2024-02-20:
https://docs.gtk.org/glib/struct.HashTable.html

Use one of the GHastable functions to get the keys

weliem, 2024-02-20:
The key is a *int and the value is a GByteArray

closed 2024-02-20 by mamigot-salto