weliem/bluez_inc #25: Handling RSSI threshold for device pair requests in peripheral mode

lizziemac, 2024-03-07:
I have a device, acting as a Bluetooth peripheral, that I would like to have reject pairing requests from central devices that are not within a certain RSSI range. Specifically, I want to block devices that have a weak signal strength from pairing, ensuring that only devices in close proximity can establish a connection.

Here's the approach I've attempted in my code:

```cpp
gboolean on_request_authorization(Device *device) {
  // Check that the RSSI is strong enough. We don't want to connect to a device
  // that isn't right next to us (for security reasons)
  short rssi = binc_device_get_rssi(device);
  const char *device_name = binc_device_get_name(device);
  if (rssi < BLUETOOTH_RSSI_THRESHOLD) {
    LOGI("rejecting authorization for '%s' because of weak signal. RSSI: %d",
         device_name, rssi);
    return FALSE;
  }

  LOGI("granting authorization for '%s'", device_name);
  return TRUE;
}
```

However, the `rssi` value seems to default to `-255`, leading to all requests being rejected, regardless of their actual signal strength. I'm thinking that the peripheral might not be able to obtain the RSSI value of the central device before pairing, which would be the root cause of the issue. Is my current understanding correct, or is there a different method to achieve this functionality?

Thank you in advance for your insights and help!

weliem, 2024-03-10:
Hi Lizzie

When a central connects there is almost nothing known about it. The only thing you can count on is the mac address being there. So the RSSI value is the default -255. It seems that Bluez only updates RSSI values when it is discovering/scanning. But as a peripheral you typically don't scan so the RSSI value isn't there. Even when scanning, after a connection is made the RSSI value is never updated.

Of course this is a Bluez limitation because in theory it should be possible to get the RSSI for a connection. On Android for example, it is possible. You could try to request this feature to the Bluez team.

All in all, I'd say your approach is not feasible at the moment.

lizziemac, 2024-03-11:
Thanks for the response! I appreciate it. I'll close this out

closed 2024-03-11 by lizziemac