weliem/bluez_inc #27: User data on all top-level `bluez_inc` types?

charlesnicholson, 2024-04-05:
Thanks for `bluez_inc`, it's so much easier to work with than bluez / dbus :)

A lot of the `bluez_inc` callbacks just give you an `Adapter *` or a `Device *`, but in a system that manages more than one adapter / device, it's hard to know which one is which. So, I need to end up writing some globally-accessible thread-safe data structure that I can pass these pointers to in exchange for some of my application context.

If there were functions like
```
void binc_adapter_set_user_data(Adapter *adapter, void *user_data);
void *binc_adapter_get_user_data(Adapter *adapter);
```
and
```
void binc_device_set_user_data(Device *device, void *user_data);
void *binc_device_get_user_data(Device *device);
```

Then it would be trivial to load up the `bluez_inc` objects with a pointer to my application context, and retrieve it in the various `bluez_inc` callbacks.

charlesnicholson, 2024-04-05:
(Also, if the implementation is as simple as I'm making it sound, I'd happily throw together a PR for this)

weliem, 2024-04-06:
Feel free to raise a PR!

closed 2024-04-09 by charlesnicholson