weliem/bluez_inc #39: Advertising, scan and connection parameters

mamigot-salto, 2024-04-26:
Hi! In order to make our application faster and improve the user experience, I'd like to change the default advertising, scan and connection parameters (mainly I'd like to change advertising interval and connection interval) but I'm not able to find the functions in the library to do so.
Are there functions currently in the library to set these paremeters? If not, do you have plans to add this feature in a near future?
Thanks a lot

weliem, 2024-04-26:
There are no DBus apis to change any of these. However, you can modify some of them by editing the Bluez config file.

weliem, 2024-04-26:
https://github.com/bluez/bluez/blob/master/src/main.conf

mamigot-salto, 2024-04-29:
Understood. Thank you!

closed 2024-04-29 by mamigot-salto