weliem/bluez_inc #40: Add Device into characteristic callbacks?

bojennett, 2024-04-29:
On iOS, the peripheral delegate callbacks (didUpdateNotificationStateFor, didUpdateValueFor, didWriteValueFor, didUpdateValueFor) contain the peripheral object.  Same for Android on the BluetoothGattCallback (onCharacteristicChanged, onDescriptorWrite, onCharacteristicWrite, onCharacteristicRead).

It's nice having this device object in the callbacks, that way you don't have to keep your own reference around to it.  As an example, after getting something back that you read, or receiving a specific notification status, you might want to disconnect because you have everything you need.  Having the device object right there in the callback makes that easy as opposed to having to hold it yourself in your code and then free it after.  I agree that you can have your own callbacks for each device when you set them on connection so you can "know" who is who, but you still need to hold onto that reference.

weliem, 2024-04-29:
Makes total sense! I did that already with several other callbacks but indeed not with the characteristic ones...not sure why I didn't

weliem, 2024-04-29:
Fixed by d214565f00e4c91c7ba7bca34f7c9e8a303576ba

closed 2024-04-29 by weliem