weliem/bluez_inc #46: Issue with scanning after turning bluetooth of and then back on

bojennett, 2024-06-05:
I have an app that acts as central.  When it starts up, it checks the bluetooth state, and if it's off, turns it on and then once on, starts scanning.  I have scanning running continuously.  For example, a device may get connected, but the scanner is still running such that if the device disconnects, it can be found again and reconnect.  As all this happens, I see as I'm starting scanning that I am getting the 'starting' and 'started' states from `AdapterDiscoveryStateChangeCallback` as expected.

if while this app is running, I go through another method to turn bluetooth off, I get the `AdapterPoweredStateChangeCallback` as expected with the state off.  I try to use this to turn scanning off, but since the adapter is not powered, this calls fails (not surprising).

If I then go into this external method of toggling bluetooth, and I toggle it back on, I get the `AdapterPoweredStateChangeCallback` as expected saying the state is ON, and then I try to start scanning again.

However, it never seems to actually start scanning again.  I don't get any of the `AdapterDiscoveryStateChangeCallback` anymore.  I tried turning scanning off once I see the power state to on, and then turning scanning on, but it doesn't seem to work.  It's as if I cannot scan anymore.

bojennett, 2024-06-05:
The way I did this - I have one terminal running my app, and then in the other terminal I do `bluetoothctl power off` to then see my app say "hey power is off", and then I wait a few seconds and in this other terminal do `bluetoothctl power on` to turn it back on, and I see that the power did come back on in my app, and I call the scanner, `binc_adapter_set_discovery_filter` and `binc_adapter_start_discovery`, but nothing seems to happen.

weliem, 2024-06-06:
I think this is a bug in my library. I need to reset some internal state when bluetooth is switched off...

Will pick this up

weliem, 2024-06-07:
fixed by ec31ca6650d8f378b2161dbb09a0d8d20cfb2105

closed 2024-06-07 by weliem