weliem/bluez_inc #47: Reconnect to a previous device causes strange behavior

bojennett, 2024-06-05:
As a central, I'm seeing a weird behavior, as if either binc (or more likely bluetoothctl) is "remembering" something about a previous device.

A device will disconnect, but the scanner is still running, and so as soon as the device comes back into range, it will connect again.  However, when this happens, before I've done any configuring of the device, I will get a few notifications (two from the battery, two from one of my custom characteristics).  I can see from btmon that these get produced from bluetooth, but I also see that these die NOT come from my device.  Separate logging on my device shows no activity there.

Then, when I go to configure my newly connected device, I go to turn on the battery level characteristic to send notifications, but (1) the library is telling me this characteristic already has notifications enabled (`binc_characteristic_is_notifying` saying true), and then when I try to enable it anyway via `binc_characteristic_start_notify`, the command goes out, but there is no log saying it returned (no `notifying true <uuid>`) probably because it's already true so it didn't "change"?

bojennett, 2024-06-05:
My workaround for this at the moment is that I check to see the state before I do an action.  If I want to turn notifications on, but it's already on, I say "hey that's weird" (because I know in my application I never want to turn it off).  I will then force it off, and when I see that it was off I know that since I never tried to turn it off, I should try to turn it on.  It's a hack because there is the possibility I could get into a loop here maybe, but it suffices.

But I'm curious if you are keeping objects around in your library based upon the UUID of the characteristic, and they are staying even if a device disconnects?  And then the internal state of that is "notifications were enabled".  Or, as I said, most likely it's a bluetoothctl thing

weliem, 2024-06-06:
This is probably expected behavior. Bluez retains notification state for as long as the object is on the DBus. For paired devices this is the correct behavior and in line with the bluetooth standard. But for non-paired devices this should not be done. What I do in my apps is to remove non-paired devices from Dbus upon disconnecting. That way Bluez will 'forget' the notification state...

bojennett, 2024-06-06:
Got it.  This is a non-paired device (BINC_BOND_NONE when discovered, and no pairing request made).  Is there some way to "force" something off the DBUS when it's disconnected?

My workaround is fine.  If you think about it, I can't think of a reason why I would have notifications enabled and would want to turn it off, so if I were to ever see it on when I think I want to turn it on, it's easy enough to assume that maybe the data is stale and to force it off, and then back on.

bojennett, 2024-06-06:
Boom!  That did the trick.

I'm wondering... since this is DBUS related, would there be any benefit to also "hiding" this?  i.e. if we sense a device is disconnected, and it isn't a paired device, could adapter/device just internally call the remove?

Totally get why you might be reluctant to do that.

weliem, 2024-06-07:
Well, if you remove it always, then you always need to discover it again before you can connect again. This might not be desirable in all use cases...e.g. if you want to connect to known devices quickly

closed 2024-06-07 by weliem

Srikor, 2024-09-03:
So, how do I connect to already paired devices? I'm a newbie and I can't see the paired devices in discovery and I'm not getting any connection change event when the paired device is powered on.