weliem/bluez_inc #48: Peripheral - Device Information Service

JoeDexcom, 2024-07-15:
I built the basic example project and connected it to a generic scanner.

When connected, it is finding a device information service, with one characteristic, the PnP ID.  I'm building this on an PC running Ubuntu.

Is the device information service added automatically because that's what the PC provides?  I can't seem to add a device information service myself with any characteristics.

The project I'm trying to build is a device that emulates another device (as a POC, the example I'm building is a Nonin 3230 pulse oximeter).  So I want to mimic all of the services and characteristics, including the device information service.

But I can't seem to add my own, and don't know what to do about the one that seems to be there.

weliem, 2024-07-15:
A similar is has already been reported on this. Yes, Bluez adds the Device Information service itself. I haven't yet found a way to prevent this or add things to it...

weliem, 2024-08-25:
You can disable the DIS service in `/etc/bluetooth/main.conf`

Add `DeviceID = false` to prevent Bluez from adding its own DIS service

closed 2024-08-25 by weliem

JoeDexcom, 2024-08-26:
Excellent!  Looks like i had to reboot the machine to make it work.  I tried doing bluetootctl power off/power on, but then what happened is it created TWO device information services (the one it used to have with the PnPID, and then the one i created).

Maybe i didn't have to reboot, but rebooting did it.  There must be a cache or something somewhere that held onto the one it used to create on its own.