weliem/bluez_inc #57: Ubuntu 22.10: Discovering starts only after disable and enable the bluetooth adapter

freibauer, 2024-09-04:
I am using the example and adapted the filters for my device.
If the bluetooth adapter is disabled the example is not starting.
If the bluetooth adapter is enabled the example is starting the discovering but nothing
happens. If i disable and enable the adapter (with the Ubuntu-GUI) during discover 
the discovering is starting and the example is pairing.
Perhaps rfkill needs to be used in the correct moment, but i am not sure when.

martijnvanwelie, 2024-09-04:
Please share relevant code and log.

weliem, 2024-09-04:
try `rfkill unblock bluetooth` before running your app.....

freibauer, 2024-09-04:
Many thanks for the quick replies. I get a start-script with the unblock to work.
Which looks like:
sudo hciconfig hci0 down; sudo rfkill unblock bluetooth ; sudo hciconfig up ; sudo ./central

closed 2024-09-06 by weliem