weliem/bluez_inc #58: Gatt exchange mtu event not work

Gearspre, 2024-09-08:
I am trying to use this lib to create gatt server, it works. But no matter how I send the exchange MTU event, its MTU always remains at 517. 
And I find in **application.c line:1035**  has one todo : *deal with the offset & mtu parameter*
So is this a known issue, and is there any simple solution?

weliem, 2024-09-09:
I have been looking into this but I don't think Bluez has an API for this.

Gearspre, 2024-09-09:
Yes. I found a link discussing this thing : [Determine ATT MTU size over DBUS](https://lore.kernel.org/all/-JmTPP9qq2ljmysdKs_4nDqzQMZH4NMScqdEFmY7eo4Eu5GDniGqk-lMiLXCo7O1RZgB4BI_UsKZS0KdET69HZksuaRIKE_0mrQJS885K-M=@krauser.org/T/)

Now I have another problem. How can I get the negotiated MTU size from GattCharacteristic? 
A simple solution is get from ReadValue or WriteValue method options. But it's not a good solution. Do you have any idea?

weliem, 2024-09-10:
Using the MTU from the ReadValue and WriteValue options parameter is totally fine to use. However, that doesn't enforce a MTU. It simply says what the current MTU is....

closed 2025-05-05 by weliem