weliem/bluez_inc #60: discovery filter with beacons or uuids?

abqjln, 2024-09-10:
My application requires the central to both connect with multiple servers that advertise uuids, and also to continuously scan for specific beacons that do not advertise uuids and are not connectable. My system works if I just specify NULL for both uuid and pattern, but I'd like to filter at a lower level if possible.

Currently, if a pattern is specified in binc_adapter_set_discovery_filter, the function gboolean_matches_discovery_filter returns FALSE if it does not find either the device name or address. 

Would these ideas be generally useful to incorporate (they are to me)?

1) Can the match filter be an OR between a pattern match and the uuid array index? The beacon pattern will repeatedly be there since we don't connect (where the server should stop advertising),.

2) Beacons often have null names and random addresses. Beyond name and address, is it possible to search for type flags or manufacturer ID or even better a "unique" series of bytes in the packet?

3) Could the pattern match also be expanded to be an array of patterns, e.g., detect both Apple ibeacon or Google Eddystone but not Samsung?

Thanks for considering this!

weliem, 2024-09-22:
Bluez doesn't support any other filtering than service uuid or name. This library hence doesn't do anything else either.

abqjln, 2024-09-24:
It seems I need to dig deeper into bluez to understand how to obtain/filter the raw advertising packet from a beacon sans name, fixed address, or uuids. BTW, this library is quite enabling for my work, thanks.

closed 2024-09-24 by abqjln