weliem/bluez_inc #62: bluez_inc doesn't compile on pi zero

jfclere, 2024-09-23:
[  6%] Building C object binc/CMakeFiles/Binc.dir/parser.c.o
/home/pi/bt/bluez_inc/binc/parser.c: In function ‘parser_get_string’:
/home/pi/bt/bluez_inc/binc/parser.c:266:63: error: conversion to ‘gssize’ {aka ‘int’} from ‘unsigned int’ may change the sign of the result [-Werror=sign-conversion]
                             (unsigned int) parser->bytes->len - parser->offset);
                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [binc/CMakeFiles/Binc.dir/build.make:167: binc/CMakeFiles/Binc.dir/parser.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:136: binc/CMakeFiles/Binc.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

Work-around:
pi@raspberrypi:~/bt/bluez_inc $ git diff .
diff --git a/binc/parser.c b/binc/parser.c
index 2ad4b7c..04554b3 100644
--- a/binc/parser.c
+++ b/binc/parser.c
@@ -263,7 +263,7 @@ GString *parser_get_string(Parser *parser) {
     g_assert(parser->bytes != NULL);
 
     return g_string_new_len((const char *) parser->bytes->data + parser->offset,
-                            parser->bytes->len - parser->offset);
+                            (int) (parser->bytes->len - parser->offset));
 }
 
 GDateTime *parser_get_date_time(Parser *parser) {

weliem, 2024-09-23:
i guess it has nothing to do with the pi zero? so just that extra cast needed?

jfclere, 2024-09-24:
correct it has nothing to do with pi zero, but with 32 bits OS and gcc being pick with the conversion between integers.

weliem, 2024-09-25:
Fixed by 6017fbdd246e882217fe0de6bc7776b7f31b513a

closed 2024-09-28 by weliem