weliem/bluez_inc #65: binc_create_application works on first, but always fails on second Bluetooth adapter

abqjln, 2024-09-27:
Is the issue the re-use of path /org/bluez/bincapplication? Can it be made unique? Am I thinking about this wrong? Thank you.

I am attempting to create two separate BLE servers on hci0 and hci1 using RPI4 (bt 5.78, Debian12, arm64). Get the same errors on RPI5 (bt 5.66, Debian12, arm64). The code sets either hci0 or hci1 (external Dual-mode USB adapter Edimax USB2.0 Bluetooth 5.0 + EDRinstalled). The first server instance (either hci0 or hci1) always succeeds but the second one always fails, regardless of the order.

(My use of p_server_s->xxx in the pseudocode creates separate instances of the various variables, i.e., they are not common to the two instances.)

Pseudocode:
` p_server_s->dbusConnection = g_bus_get_sync( G_BUS_TYPE_SYSTEM, NULL, NULL );`
` p_server_s->adapter = binc_adapter_get(  p_server_s->dbusConnection, "hci0" | "hci1" );`
At this point  the adapters share a dbusConnection (singleton nature??) but have different addresses
[server_init (my code)] Server0: dbusConnection=0x1e3d000 is using adapter hci1 [08:BE:AC:32:75:24]
[server_init (my code)] Server1: dbusConnection=0x1e3d000 is using adapter hci0 [D8:3A:DD:47:F5:57]

Case 1) If create the application next (my order),
` p_server_s->app = binc_create_application(  p_server_s->adapter );`
[Application] failed to publish application
[Application] failed to publish local service  (tries since don't trap error on binc_create_application)
[Application] Error An object is already exported for the interface org.freedesktop.DBus.Properties at /org/bluez/bincapplication/service0

Case 2) If advertise next (example/peripheral.c order)
`binc_advertisement_set_local_name( p_server_s->advertisement,  p_server_s->name );`
`binc_advertisement_set_services( p_server_s->advertisement,  p_server_s->adv_uuids );`
`binc_adapter_start_advertising( p_server_s->adapter,  p_server_s->advertisement );`
[Advertisement] registering advertisement failed: An object is already exported for the interface org.bluez.LEAdvertisement1 at /org/bluez/bincadvertisement
[Application] failed to publish application
[Application] failed to publish local service  (tries since don't trap error on binc_create_application)
[Application] Error An object is already exported for the interface org.freedesktop.DBus.Properties at /org/bluez/bincapplication/service0

abqjln, 2024-09-28:
Oh no. I hope I'm not digging a one-off rat hole!  Ultimately I need multiple servers each with different addresses. A single adapter would be best, but I am having difficultly understanding/getting the private random address function with my client that I do not control.

 So I thought two adapters with different public addresses would be an easy way to test... Please don't pursue this if not more widely useful. Thank you.

weliem, 2024-09-28:
I randomized the paths for application and advertisement, see 835945be49d531b76432d956a4106e0174d5a803.

Let me know if that solves your issues

abqjln, 2024-09-28:
Now get an application registration error with one or  both adapters. The applications get published, then publish services and local characteristics, but then after a timeout give

2024-09-28 08:58:36:767 DEBUG [Application] adding /org/bluez/bincapp_hci0_Hjqe/service0
2024-09-28 08:58:36:768 DEBUG [Application] adding /org/bluez/bincapp_hci0_Hjqe/service0/char4
...bunch of them
2024-09-28 08:58:36:770 DEBUG [Advertisement] setting advertising MinInterval to 200ms (requires experimental if version < v5.77)
2024-09-28 08:58:36:770 DEBUG [Advertisement] setting advertising MaxInterval to 500ms (requires experimental if version < v5.77)
2024-09-28 08:58:36:770 DEBUG [Adapter] failed to register application (error 36: GDBus.Error:org.bluez.Error.Failed: No object received)

weliem, 2024-09-29:
Any more info you can give me? Some code perhaps? I am struggling to understand what is going wrong...

abqjln, 2024-09-29:
Sorry my error. Confirmed that your changes work fine for both adapters, including start/stop advertising independently. I had inserted a delay for debugging that was causing the timeout.  Use case still not working FWIW Garmin watches.

abqjln, 2024-09-29:
Another related issue I think. Nothing proprietary about my code below, I will publish when it works.

I can now register two application/adapter/advertisements independently. But upon the first connection from client device, the centralStateCallback gets erroneously triggered for both the first and second adapters. Same is true for disconnects--disconnecting the device on one adapter also disconnects the other.

In adapter.c,
`typedef void (*RemoteCentralConnectionStateCallback)(Adapter *adapter, Device *device);`

In device.c
`struct binc_device`

both contain the adapter and device, but the binc_internal_device_XXX does not use both when acting. I think we need the device+adapter combination??

My initialization code
`int server_init( struct server_s_struct *p_server_s, char *adapter_name )
{
	if( ( p_server_s->dbusConnection = g_bus_get_sync( G_BUS_TYPE_SYSTEM, NULL, NULL ) ) == NULL ){
		log_error( TAG, KRED"Failed to get DBus connection"KNRM );
		return( -1 );
	}

	if( adapter_name == NULL ){
		log_error( TAG, KRED"NULL adapter name"KNRM );
		return( -1 );
	}

	if( ( p_server_s->adapter = binc_adapter_get( p_server_s->dbusConnection, adapter_name ) ) == NULL){
		log_error(TAG, KRED"Adapter not found"KNRM );
		g_object_unref( p_server_s->dbusConnection );
		return( -1 );
	}

	binc_adapter_set_powered_state_cb( p_server_s->adapter, &on_server_powered_state_changed );
	if( !binc_adapter_get_powered_state( p_server_s->adapter ) ){
		binc_adapter_power_on( p_server_s->adapter );
		log_debug( TAG, DCOLOR"Powered on adapter '%s'"KNRM, binc_adapter_get_path( p_server_s->adapter ) );
	}

	if( (p_server_s->app = binc_create_application( p_server_s->adapter )) == NULL ){
		log_error( TAG, KRED"Failed to create application" );
		return( -1 );
	}
	log_debug( TAG, DCOLOR"%s: Published application %s using dbusConnection=%p on adapter %s [%s]"KNRM,
			p_server_s->name, binc_application_get_path( p_server_s->app ), p_server_s->dbusConnection, \
			binc_adapter_get_name( p_server_s->adapter ), binc_adapter_get_address( p_server_s->adapter ) );

	//******************************************************************
	// GATT added automatically
	// Initialize DIS and BS (not advertised, discovered)
	service_dis_init( p_server_s->app, &p_server_s->dis_s );
	//service_dis_inspect( &p_server_s->dis_s );
	service_bs_init( p_server_s->app, &p_server_s->bs_s );
	g_timeout_add( BS_NOTIFY_MS, server_notify_bl, p_server_s );

	//******************************************************************
	// Additional services based on adv_uuids
	for( guint j = 0; j < p_server_s->adv_uuids->len; j++ )
	{
		if( g_str_equal( g_ptr_array_index( p_server_s->adv_uuids, j ), HRS_SERVICE_UUID ) ){
			service_hrs_init( p_server_s->app, &p_server_s->hrs_s );
			g_timeout_add( HRS_NOTIFY_MS, server_notify_hrm, p_server_s );
		}

		if( g_str_equal( g_ptr_array_index( p_server_s->adv_uuids, j ), CPS_SERVICE_UUID ) ){
			service_cps_init( p_server_s->app, &p_server_s->cps_s );
			g_timeout_add( HRS_NOTIFY_MS, server_notify_cpm, p_server_s );
		}

		if( g_str_equal( g_ptr_array_index( p_server_s->adv_uuids, j ), CSCS_SERVICE_UUID ) ){
			service_cscs_init( p_server_s->app, &p_server_s->cscs_s );
			g_timeout_add( HRS_NOTIFY_MS, server_notify_cscsm, p_server_s );
		}

	}

	//******************************************************************
	// Advertising
	// Local name to be advertised along with uuids
	// Garmin does not detect name, only address, but nRF Connect does
	p_server_s->advertisement = binc_advertisement_create();
	binc_advertisement_set_local_name( p_server_s->advertisement, p_server_s->name );
	binc_advertisement_set_services( p_server_s->advertisement, p_server_s->adv_uuids );
	binc_adapter_start_advertising( p_server_s->adapter, p_server_s->advertisement );

	//******************************************************************
	// Set callbacks
	binc_application_set_char_read_cb( p_server_s->app, &on_server_local_charact_read );
	// until pull request binc_application_set_char_updated_cb( p_server_s->app, &on_server_local_charact_updated) ;
	binc_application_set_char_write_cb( p_server_s->app, &on_server_local_charact_write );
	binc_application_set_desc_write_cb( p_server_s->app, &on_server_local_desc_write);
	binc_application_set_char_start_notify_cb( p_server_s->app, &on_server_start_notify_from_client );
	binc_application_set_char_stop_notify_cb( p_server_s->app, &on_server_stop_notify_from_client );

	// Setup remote central connection state callback
	binc_adapter_set_remote_central_cb( p_server_s->adapter, &on_adapter_client_state_changed );
	binc_adapter_register_application( p_server_s->adapter, p_server_s->app );

return( 0 );
}
`
and

`void on_adapter_client_state_changed( Adapter *adapter, Device *client )
{
	char *deviceToString = binc_device_to_string( client );

	ConnectionState state = binc_device_get_connection_state( client );

	if (state == BINC_CONNECTED) {
		log_debug(TAG, KGRN"Client '%s' [%s] connected to [%s]"KNRM, \
				binc_device_get_name( client ), binc_device_get_address( client ), binc_adapter_get_address( adapter ) );
		binc_adapter_stop_advertising( adapter,  binc_adapter_get_advertisement( adapter ) );
	}
	else if (state == BINC_DISCONNECTED){
		log_debug(TAG, KRED"Client '%s' [%s] disconnected from [%s]"KNRM, \
				binc_device_get_name( client ), binc_device_get_address( client ), binc_adapter_get_address( adapter ) );
		binc_adapter_start_advertising( adapter, binc_adapter_get_advertisement( adapter ) );
	}
	else if (state == BINC_CONNECTING) {
		// Wait
	}
	else {
		log_error( TAG, KRED"Client '%s' [%s] has an indeterminate connection state to [%s]"KNRM, \
				binc_device_get_name( client ), binc_device_get_address( client ), binc_adapter_get_address( adapter ) );
		// Disconnect and restart
		binc_device_disconnect( client );
		binc_adapter_start_advertising( adapter, binc_adapter_get_advertisement( adapter ) );
 	}

	g_free(deviceToString);
}
`

abqjln, 2024-09-30:
Narrowing down...The binc_adapter_set_remote_central_cb is getting triggered twice, seemingly because the adapter path is changing from an incorrect one (in this case hci1) to correct (hci0). The device path is correct in both instances.

Set up callback
`binc_adapter_set_remote_central_cb( p_server_s->adapter, &on_adapter_client_state_changed );
`

Callback
`void on_adapter_client_state_changed( Adapter *adapter, Device *client )
{
	log_debug( TAG, KYEL"State change: (Adapter: path=%s) (Device: path=%s) "KNRM,\
			binc_adapter_get_path(adapter), \
			binc_device_get_path( client ) );
...
`

Logs
(1st) DEBUG [on_adapter_client_state_changed] State change: AdapterPath=/org/bluez/hci1, DevicePath=/org/bluez/hci0/dev_E0_48_24_50_BB_EF

(2nd) DEBUG [on_adapter_client_state_changed] State change: AdapterPath=/org/bluez/hci0, DevicePath=/org/bluez/hci0/dev_E0_48_24_50_BB_EF

The adapter path (hci1--wrong) does not match the device path (hci0) on the first cb, but does on the cb (hci0).

weliem, 2024-09-30:
ok thanks, closing in on the real issue. Need to study the code to understand why this is happening....

abqjln, 2024-10-01:
Closing further. In the first invocation of binc_internal_device_appeared, the user_data (cast to Adapter*) contains an incorrect path to the adapter. This seems to trigger a second appearance with the correct adapter path.

Past this point, the paths all seem consistent.

Discovered using

`log_debug(TAG,"\x1B[33m%s: Created device using object=%s, adapterpath=%s\x1B[0m",__func__,object,binc_adapter_get_path(adapter));`

 into blank line 334 in adapter.c (escape characters make it yellow)

First try (HRS device at hci0):
[Adapter] binc_internal_device_appeared: Created device using object=/org/bluez/hci0/dev_E0_48_24_50_BB_EF, adapterpath=/org/bluez/hci1
...connects to mismatched adapter/device

Second try:
[Adapter] binc_internal_device_appeared: Created device using object=/org/bluez/hci0/dev_E0_48_24_50_BB_EF, adapterpath=/org/bluez/hci0
...connects to matched adapter/device, but it's messed up and notifications do not work

*****************
Related??  Noticed that g_variant_iter_init is used in line 339, 595, 602, 623, but not before iteration loops in lines 214, 300, 331, 389, 443, 590. Still new to GLib so do not quite understand when init is necessary?

Do you need an external adapter to test--will send you one.

weliem, 2024-10-01:
I see indeed a case where a newly discovered device can be assigned the wrong adapter. Add an extra check in ee63c0447377c551cd41cebfe74f199b3c39ecf9

Can you try to see if that helps? I have more adapters but need to dig them up....

abqjln, 2024-10-02:
Further...has to update my client side this time to handle the progress.

Now it throws
Bail out! ERROR:/mnt/code/bluez_inc_jln/binc/device.c:946:binc_device_get_connection_state: assertion failed: (device != NULL)

calling from binc_internal_device_changed line 448 in adapter.c

Looks like g_hash_table_lookup is returning a NULL device but not also satisfying g_str_has_prefix condition in line 442. Another check?

weliem, 2024-10-02:
Damn, I did that check wrong......fixed now by cd77046e4a532cb1b00f18c8c2abea702b8392f0

I did some test with my 2nd adapter and it is working for me now

abqjln, 2024-10-03:
Works! Server side has one application with  two different adapters, each with different services and different characteristics tested with read and notify.  Thanks.

closed 2024-10-03 by abqjln

weliem, 2024-10-03:
Nice! These were all good improvements that made this library better when dealing with multiple adapter.