weliem/bluez_inc #67: Feature request: logger TAG and colors

abqjln, 2024-10-07:
Digging deep into this great library. Two logging mods I've been manually making might be broadly useful. 

1) Replace TAG definitions with hardcoded source filename.c with __func__. The functions have the file name embedded so logging their name gives more specific info and still the source file info. 

2) Would it be possible to set colors depending on the log level. That way ERROR, WARN, DEBUG, and INFO would stand out in the stream. I am doing this manually so it works with Windows ssh terminals and gcc.

e.g., 
#define LOG_NRM  "\x1B[0m"
#define LOG_ YEL  "\x1B[33m"
#define LOG_RED  "\x1B[31m"
#define LOG_GRN  "\x1B[32m"
#define LOG_BLU  "\x1B[34m"
#define LOG_MAG  "\x1B[35m"
#define LOG_CYN  "\x1B[36m"
#define LOG_WHT  "\x1B[37m"

#define LOG_ERROR LOG_RED

These would be prepended and postpended to the format statement. Color could be specified redefining:

log_set_level(LogLevel(level, ERROR_COLOR, DEBUG_COLOR, WARN_COLOR, INFO_COLOR );
Thanks.

valad47, 2024-10-27:
I think it would be a better choice to set color with function like `log_set_level_color(LogLevel level, unsigned char color)`, where color would be smth from [that table](https://gist.github.com/fnky/458719343aabd01cfb17a3a4f7296797#256-colors) or rgb version

lizziemac, 2024-11-05:
If this gets implemented, can we ensure that the logs can also be uncolored (preferably by default)? Some log collectors/viewers don't support colors well and they just show up as raw characters

abqjln, 2024-11-15:
I submitted a PR to make the TAG declarations uniform across the library, so I just run `sed -i '/static const char \*const TAG/c\#define TAG __func__' binc/*.c` on my fork and use gcc. Eliminates the type checking but I find it extremely helpful to know the specific function.

The escapes in the logs mentioned @lizziemac are indeed yucky. Standing down on that one, will do it manually since I compile from scratch anyway. Thanks--this is an enormously useful library.