weliem/bluez_inc #71: binc or not? Automatic notification starts/stops for battery service : char <0000180f>:<00002a19>

abqjln, 2024-10-13:
I observe independent (not in my control) start/stop notifies of battery level. Are these built into binc or maybe a feature of bluez?

I've implemented three servers with notify characteristics (<00001814>:<00002a53> RSCS, <00001818>:<00002a63> CPS, <0000180d>:<00002a37> HRS) on three different adapters/applications on Rpi4 (bluez 5.78).  My client  (Rpi5, 5.78) controls the proper start/stop of these notifies, but my servers get unwanted battery level  notification starts commands shortly after connecting, and notification stop commands after disconnection. Not something I am doing.

I couldn't find it in binc--did I miss something? A default keep-alive or something?

weliem, 2024-10-14:
Bluez uses the Battery plugin by default. You can disable it by adding a startup option:

bluetoothd -P battery

To do that you must edit bluetooth.service

abqjln, 2024-10-14:
Works, thanks.  Now that I've learned the term "plugin" I found more background why this behavior exists.

https://stackoverflow.com/questions/66690927/ios-central-with-bluez-peripheral-disconnect-due-to-insufficient-auth
https://lore.kernel.org/linux-bluetooth/CAG-pW8Ev_ntv1=o0ndd_2FXTdbsvHJ=FVsB-YXUucVtyKSO0Sg@mail.gmail.com/t/

closed 2024-10-14 by abqjln

abqjln, 2024-10-14:
Duh, also just found this in the wiki, sorry.

Separately, can I buy you a coffee? Greatly appreciate how binc has enabled
my projects and your diligence in maintaining it.

weliem, 2024-10-17:
Let me know whenever you are in Amsterdam and you can buy me a coffee!

abqjln, 2024-10-19:
Will be there Sunday 22 June 2025 somewhere between a bus and a cruise
ship...