weliem/bluez_inc #72: Executing binc_characteristic_write() multiple times raises dlib-assertion G_IS_DBUS_CONNECTION

freibauer, 2024-10-16:
Hello everybody,

i have a issue writing multiple times a characteristic.
After exactly 12 times binc_characteristic_write() the dbus throws an exception:

(process:14758): GLib-GIO-CRITICAL **: 19:12:31.273: g_dbus_connection_call_finish_internal: assertion 'G_IS_DBUS_CONNECTION (connection)' failed

I am using the binc_characteristic_write() to write data to a connected ble-modul which is 
in the u-blox SPP-mode.
 
I have used the initialization like in the examples without setting a callback:
```
    GVariant *result = g_dbus_connection_call_sync(dbusConnection,
                                                   BLUEZ_DBUS,
                                                   "/",
                                                   INTERFACE_OBJECT_MANAGER,
                                                   "GetManagedObjects",
                                                   NULL,
                                                   G_VARIANT_TYPE("(a{oa{sa{sv}}})"),
                                                   G_DBUS_CALL_FLAGS_NONE,
                                                   -1,
                                                   NULL,
                                                   &error);
```

Best regards,
Mathias

weliem, 2024-10-17:
Ok, so this issue can be closed?

freibauer, 2024-10-17:
Yes, it can be close.
Just one question about the delays. If i firing in the on_read() the next characteristic is there any
delay caused by the library or from the dbus ? The BLE-central is much quicker sending back frames.

weliem, 2024-10-17:
I am not sure I understand your question. But the library is not adding delays anywhere. In general, there is no need to add delays.

freibauer, 2024-10-17:
Many thanks for the quick reply and all your work.

closed 2024-10-22 by weliem