weliem/bluez_inc #73: Client with multiple servers only sees first connection state change from disconnecting to disconnected

abqjln, 2024-10-22:
My central/client adapter connects to three servers simultaneously.  After successful connections, if the three server applications initiate disconnection, the client `binc_device_set_connection_state_change_cb` gets triggered for all three as "disconnecting". However, the state change to "disconnected" subsequently only occurs for the first application to disconnect.

Until the servers reconnect. Then the client state changes to "disconnect" are triggered for the 2nd and 3rd servers.  All is fine then, but for that intermediate period my client does not properly hold the actual connection state, although the servers know they are disconnected.

It looks like the connection state cache(?) is getting flushed when it reconnects? Is there an fflush for the `inc_device_set_connection_state_change_cb' ? Thanks.

abqjln, 2024-10-23:
I had an printf error--you are correct. The peripherals are initiating disconnect and the client does not see a "disconnecting" state. (The client does see a "connecting state") 

But the buffering issue on the "disconnected state" persists.  Here's  debug trace to illustrate:

`Client side is a pi5 with bluez 5.78 with onboard adapter. 

[bt_adapter_init]: Configured adapter '/org/bluez/hci0'
[client_agent_init_all]: Started agent '/org/bluez/BincAgent_bZQm'
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'starting'
[client_adapter_start_discovery]: Started discovery on adapter '/org/bluez/hci0', UUIDs [NULL], adv_pattern 'NULL')
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'started'

**********
At this point I start the server code on a pi4 with bluez 5.78 that spins up 3 services on 3 different adapters 
with advertised names pi4dev-CPS, pi4dev-CSCS, pi4dev-RSCS. The discovery state changes are triggered in the "disconnected" state of the on_client_connection_state_changed--I need to keep it discovering for other connections and also for a separate set of beacons not shown here.
**********

[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CPS [08:BE:AC:32:75:24] connecting
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-RSCS [D8:3A:DD:47:F5:57] connecting
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CSCS [08:BE:AC:32:78:B3] connecting
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CSCS [08:BE:AC:32:78:B3] connected
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'stopping'
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'stopped'
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CPS [08:BE:AC:32:75:24] connected
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'starting'
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'started'

[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-RSCS [D8:3A:DD:47:F5:57] connected
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'stopping'
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'stopped'

[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'starting'
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'started'

**********
Notifications continue here (deleted) until I stop the server side which disconnects all 3 servers, then immediately get
**********

[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev #1 [08:BE:AC:32:75:24] disconnected

**********
But the other two disconnect messages do not appear here, until restart the server program, then
**********I 

[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-RSCS [D8:3A:DD:47:F5:57] disconnected
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev #2 [08:BE:AC:32:78:B3] disconnected

**********
as if we are flushing some buffer? Then software reconnects and runs as before
**********

[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CPS [08:BE:AC:32:75:24] connecting
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-RSCS [D8:3A:DD:47:F5:57] connecting
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CPS [08:BE:AC:32:75:24] connected
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'stopping'
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'stopped'
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'starting'
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-RSCS [D8:3A:DD:47:F5:57] connected
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'started'
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CPS [08:BE:AC:32:75:24] connecting
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CSCS [08:BE:AC:32:78:B3] connecting
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CPS [08:BE:AC:32:75:24] connected
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'stopping'
[on_bt_adapter_discovery_state_changed]: Adapter '/org/bluez/hci0' discovery state is 'stopped'
[on_client_connection_state_changed]: Adapter [D8:3A:DD:C9:B3:B4] and server pi4dev-CSCS [08:BE:AC:32:78:B3] connected`

weliem, 2024-10-24:
ok i get it now, but I don't know yet why this happens. The library simply reacts to events coming from Bluez...and at first sight it seems Bluez is not sending them. Need to think about this more...

abqjln, 2024-10-24:
It does seem it is bluez that is not sending. Discovered I don't need to restart the server code. 

If I start or restart bluetooth on the server side (`systemctl stop/restart bluetoohctl`, the client gets the remaining two binc disconnect messages (and the follow-on discovery restarts happen. I'll formulate a question for the bluez team. Thanks.

closed 2024-10-24 by abqjln