weliem/bluez_inc #78: Change txpower from default -255?

abqjln, 2024-11-14:
I have multiple adapters in my system and would like to change their tx power levels, preferably dynamically. Is this possible in binc even on a single adapter? I may be lost and searching for the wrong thing or in the wrong place...

I see in binc_device_create that txpower is initialize to short -255.  I had expected +/-20dbm at 0.1 max (+/-200) or something smaller.  Would a value of +80 mean +8dbm?

binc_internal_device_update_property uses the Device->txpower parameter, but don't see where this is exposed to userspace. Also not seeing where one would set the adapter.c txpower.

I found bluez device.c has a function `void device_set_tx_power` that calls g_dbus_emit_property_changed, but I am so hoping that binc can continue to isolate me from dbus commands. Thanks for any help, including pointers to the right place.

weliem, 2024-11-17:
I just gave it a big negative value so that it obvious it wasn't set. I think there is a way to set it nowadays but when I initially made the library it wasn't there I think. Need to have a look.

weliem, 2024-12-14:
I added support for setting TxPower in the Advertisement 88ace1f93e378896957adac2ef12e26c4c534eb9

weliem, 2024-12-14:
For the Device object the TxPower is read only and is derived from the advertisement. See [Device.rst](https://github.com/bluez/bluez/blob/master/doc/org.bluez.Device.rst#int16-txpower-readonly-optional)

closed 2024-12-14 by weliem