weliem/bluez_inc #80: Can the binc server issue stop notify command (to itself) on disconnect?

abqjln, 2024-12-07:
Can a binc server tell itself to stop notifications or clear it's own CCCD, upon disconnect from the central?

****
I have a binc server using bluez 5.79 connecting to a central (Garmin watch) . When the central disconnects, it triggers the connection state change callback on the server. However, the server continues to send notifications.

I certainly am confused by this behavior and would appreciate any insight. Doing some research, the TI stack has the ability to clear the CCCD
https://e2e.ti.com/support/wireless-connectivity/bluetooth-group/bluetooth/f/bluetooth-forum/557244/notification-cccd-status

and SiLabs allows stopping notification
https://community.silabs.com/s/question/0D51M00007xeQHUSA2/disable-able-notification-after-ble-disconnect?language=en_US

Thank you.

weliem, 2024-12-08:
I honestly don't know. Bluez is managing the CCCD entirely and what it does is not entirely in line with the Bluetooth spec.

abqjln, 2024-12-08:
I no longer think it is a CCCD question. Tracking down why I continue to notify even when central disconnect cb triggered. 

I am testing binc_application_char_is_notifying (characteristic->notifying) in a gloop but it is not being set FALSE on disconnection. 

Should characteristic->notifying updates on disconnection be in binc?  I'm not sure where.

Or would changing from get_local_characteristic to  characteristic_get_property in binc_application_char_is_notifying work?

Otherwise I can craft a local flag using my connection cb.
Thanks.

closed 2025-05-24 by weliem