weliem/bluez_inc #94: src/advertising.c:parse_advertisement() Error parsing SecondaryChannel property

liuguoping1024, 2025-04-02:
I found a new exception:

Runtime log:

```
# ./examples/peripheral/peripheral
2025-04-02 07:51:06:835 DEBUG [Adapter] finding adapters
2025-04-02 07:51:06:855 DEBUG [Adapter] found adapter '/org/bluez/hci0'
2025-04-02 07:51:06:858 DEBUG [Adapter] found 1 adapter
2025-04-02 07:51:06:859 DEBUG [Main] using default_adapter '/org/bluez/hci0'
2025-04-02 07:51:06:862 DEBUG [Application] successfully published application
2025-04-02 07:51:06:864 DEBUG [Application] successfully published local service 00001809-0000-1000-8000-00805f9b34fb
2025-04-02 07:51:06:866 DEBUG [Application] successfully published local characteristic 00002a1c-0000-1000-8000-00805f9b34fb
2025-04-02 07:51:06:867 DEBUG [Application] successfully published local descriptor 00002901-0000-1000-8000-00805f9b34fb
2025-04-02 07:51:06:868 DEBUG [Application] set value <68656c6c6f20746865726500> to <00002901-0000-1000-8000-00805f9b34fb>
2025-04-02 07:51:06:910 DEBUG [Application] adding /org/bluez/bincapp_hci0_tw7s/service0
2025-04-02 07:51:06:911 DEBUG [Application] adding /org/bluez/bincapp_hci0_tw7s/service0/char0
2025-04-02 07:51:06:912 DEBUG [Application] adding /org/bluez/bincapp_hci0_tw7s/service0/char0/desc0
2025-04-02 07:51:06:917 DEBUG [Advertisement] setting advertising MinInterval to 500ms (requires experimental if version < v5.77)
2025-04-02 07:51:06:918 DEBUG [Advertisement] setting advertising MaxInterval to 500ms (requires experimental if version < v5.77)
2025-04-02 07:51:06:951 DEBUG [Adapter] successfully registered application
2025-04-02 07:51:06:952 ERROR [Adapter] failed to register advertisement (error 36: GDBus.Error:org.bluez.Error.Failed: Failed to parse advertisement.)

^C2025-04-02 07:51:46:428 ERROR [Main] received SIGINT
2025-04-02 07:51:46:429 DEBUG [Application] freeing application /org/bluez/bincapp_hci0_tw7s
2025-04-02 07:51:46:430 DEBUG [Application] freeing service /org/bluez/bincapp_hci0_tw7s/service0
2025-04-02 07:51:46:430 DEBUG [Application] freeing characteristic /org/bluez/bincapp_hci0_tw7s/service0/char0
2025-04-02 07:51:46:431 DEBUG [Application] freeing descriptor /org/bluez/bincapp_hci0_tw7s/service0/char0/desc0

```

```
# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm

```

git version
```
root@localhost:~/hubv3_ble_wifi/bluez_inc# git log
commit db863fed265c586364e6de6ff99c3dc13ef04c3c (HEAD -> main, origin/main, origin/HEAD)
Author: xihua13104 <34115918+xihua13104@users.noreply.github.com>
Date:   Thu Mar 13 02:15:46 2025 +0800

    add rsi for advertisement (#93)
```

```
# bluetoothctl --version
bluetoothctl: 5.66

```

weliem, 2025-04-02:
can you try to enable experimental mode in the Bluez config?

liuguoping1024, 2025-04-03:
I test the code on different platforms: Orange Pi, NUC, PC, and the code works well on those platforms.

I think something wrong with the driver of the hardware I test before.

So just close the issue and move on.

Thank you for your reply.

weliem, 2025-04-03:
It might be the Bluez or kernel version that also plays a role here

liuguoping1024, 2025-04-08:
kernel version is 5.10.y

Close issue, thanks!

closed 2025-04-08 by weliem