weliem/bluez_inc #95: Trying to compile using Clang fails

IoIxD, 2025-04-07:
I tried to add the library via FetchContent:
```
# BlueZ helper.
include(FetchContent)

FetchContent_Declare(
  bluez_inc
  GIT_REPOSITORY https://github.com/weliem/bluez_inc.git
  GIT_TAG        main
)
FetchContent_MakeAvailable(bluez_inc)
```

But I get the following error:

```
In file included from /home/gavin/Projects/bose-connect-app-linux/src/build/_deps/bluez_inc-src/binc/agent.c:24:
/home/gavin/Projects/bose-connect-app-linux/src/build/_deps/bluez_inc-src/binc/agent.h:42:27: error: redefinition of typedef 'Agent' is a C11 feature [-Werror,-Wtypedef-redefinition]
   42 | typedef struct binc_agent Agent;
      |                           ^
/home/gavin/Projects/bose-connect-app-linux/src/build/_deps/bluez_inc-src/binc/forward_decl.h:31:27: note: previous definition is here
   31 | typedef struct binc_agent Agent;
      |                           ^
1 error generated.
make[2]: *** [_deps/bluez_inc-build/binc/CMakeFiles/Binc.dir/build.make:104: _deps/bluez_inc-build/binc/CMakeFiles/Binc.dir/agent.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/gavin/Projects/bose-connect-app-linux/src/build/_deps/bluez_inc-src/binc/device.c:571:57: error: implicit conversion changes signedness: 'gboolean' (aka 'int') to 'ConnectionState' (aka 'enum ConnectionState') [-Werror,-Wsign-conversion]
  571 |             binc_device_internal_set_conn_state(device, g_variant_get_boolean(property_value), NULL);
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [_deps/bluez_inc-build/binc/CMakeFiles/Binc.dir/build.make:160: _deps/bluez_inc-build/binc/CMakeFiles/Binc.dir/device.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:177: _deps/bluez_inc-build/binc/CMakeFiles/Binc.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

Setting the compiler to GCC fixes this. However it'd be really nice if I could use clang as it provides many benefits over the former.

IoIxD, 2025-04-08:
I instantly realized this repo doesn't support FetchContent. However, importing the repo as a subdirectory does not change anything

weliem, 2025-04-08:
Looks like I am unintentionally redefining a typedef. I'll have a look this evening

weliem, 2025-04-08:
Should be fixed by 72acb073270bea4b769439bb2da15543f118c0b1

Please try again

closed 2025-04-25 by weliem