weliem/bluez_inc #96: Failed to Start Advertising

pavankn, 2025-04-09:
In the peripheral example , i just changed the value of HTS_SERVICE_UUID to something else , for ex: 
#define HTS_SERVICE_UUID "3d400fb8-6339-47c8-8fb7-03fe20fa48e5"
and it just  it fails, am i missing something ??

root@nitrogen8mm:/etc# ./peripheral
2025-04-09 13:42:38:921 DEBUG [Adapter] finding adapters
2025-04-09 13:42:38:925 DEBUG [Adapter] found adapter '/org/bluez/hci0'
2025-04-09 13:42:38:925 DEBUG [Adapter] found 1 adapter
2025-04-09 13:42:38:925 DEBUG [Main] using default_adapter '/org/bluez/hci0'
2025-04-09 13:42:38:926 DEBUG [Application] successfully published application
2025-04-09 13:42:38:926 DEBUG [Application] successfully published local service 3d400fb8-6339-47c8-8fb7-03fe20fa48e5
2025-04-09 13:42:38:926 DEBUG [Application] successfully published local characteristic 00002a1c-0000-1000-8000-00805f9b34fb
2025-04-09 13:42:38:926 DEBUG [Application] successfully published local descriptor 00002901-0000-1000-8000-00805f9b34fb
2025-04-09 13:42:38:926 DEBUG [Application] set value <68656c6c6f20746865726500> to <00002901-0000-1000-8000-00805f9b34fb>
2025-04-09 13:42:38:947 DEBUG [Application] adding /org/bluez/bincapp_hci0_ee11/service0
2025-04-09 13:42:38:947 DEBUG [Application] adding /org/bluez/bincapp_hci0_ee11/service0/char0
2025-04-09 13:42:38:947 DEBUG [Application] adding /org/bluez/bincapp_hci0_ee11/service0/char0/desc0
2025-04-09 13:42:38:950 DEBUG [Advertisement] setting advertising MinInterval to 500ms (requires experimental if version < v5.77)
2025-04-09 13:42:38:950 DEBUG [Advertisement] setting advertising MaxInterval to 500ms (requires experimental if version < v5.77)
2025-04-09 13:42:38:979 DEBUG [Adapter] successfully registered application
2025-04-09 13:42:38:984 ERROR [Adapter] failed to register advertisement (error 36: GDBus.Error:org.bluez.Error.Failed: Failed to register advertisement)

weliem, 2025-04-09:
Does the original code work for you?

pavankn, 2025-04-09:
Yes the original works but why does it fail when i change my service UUID ??

abqjln, 2025-04-09:
FWIW I had a similar error and found the error was a too-long advertising packet length. I did not solve the binc issue (with a check or otherwise) but my workaround was to limit to one 16 bit UUID. The new, random UUID may be forcing a full 128-bit UUID and throwing the error?

closed 2025-05-24 by weliem