weliem/bluez_inc #97: How to invalidate attributes cache on GATT client?

xihua13104, 2025-04-10:
Hi, first, thanks for your great work on this project. It really help me a lot on learning Bluez.

We implemented a GATT server using this library, we observe that Android phones sometimes skip full GATT service discovery upon reconnection. This leads to incomplete discovery of available services and characteristics, resulting in unexpected behavior in applications relying on full service availability.

On the first connection and pairing, the Android phone properly initiates a full GATT service discovery and obtains all services and characteristics exposed by the DUT. On subsequent connections (without unpairing), the Android phone sometimes only performs partial discovery or skips service discovery entirely. We suspect this is caused by Android caching the GATT database from the previous session.

So is there any way to invalidate attributes cache on GATT client(Android phone)?

weliem, 2025-04-10:
Your question is not about this library it seems. But yes, on Android it is possible to clear the services cache. The method is called 'refresh' but you'll need to use reflection to call it. You can find an implementation in my Blessed-Android library.

xihua13104, 2025-04-11:
Thanks for the suggestion. But I mean to do it on the DUT side, not on the Android phone side. I found that the service changed characteristic under generic attribute service can be used to notify the Android phone that the GATT database on the DUT side has changed, but I can't seem to control it, it's taken over by Bluez.
Do you know how to control the service changed characteristic on Binc library?

weliem, 2025-04-11:
No idea to be honest. Never tried.

abqjln, 2025-04-13:
Agreed, thanks for the great binc library. Here's some work on the iOS version of this issue.

I have a similar need to tell iOS clients to refresh due to changes on my DUT/server. My workaround is to use iOS Settings to both forget the devices and then stop/start bluetooth. I believe the latter clears the cache--it certainly changes the MAC.

I know my binc server can send indicates from service/chars created using binc_add_service or binc_add_characteristic.

But,

- Adding binc 2a05 without first adding a binc 1801 gives a `service 00001801-0000-1000-8000-00805f9b34fb does not exist`.

- Adding both binc 1801/GATT and then binc 2a05/SCC throws no errors, but the 2a05 characteristic does not appear on the nRF scanner. I believe the binc 180a replaces the bluez original as it breaks stuff-I get application registration issues `error 36: GDBus.Error:org.bluez.Error.Failed: Failed to create entry in database` in binc_internal_register_appl_cb.

It seems binc is adding to the existing bluez 1801/GATT tree? If feels similar to the situation with 180a (DIS), where adding DeviceID=false to /etc/bluetooth/main.conf prevents bluez from creating 180a/DIS and consequently allows one to create a binc DIS. But replacing bluez 1801 with binc 1801 is more complex.

@weliem -- Do you see obstacles in directly using dbus commands to bypass binc and just have bluez send the indicate 1801/2a05?

weliem, 2025-04-14:
You can always talk directly to dbus and bypass Binc. I think I have seen a recent commit to Bluez that will allow you to get hold of the right characteristic...will look it up

abqjln, 2025-04-25:
@xihua13104 --FWIW since @welliem found that 1801 is not exposed, I tried and I was able to add 0x2a05 service changed characteristic to each service separately. Have not tested functionality, but am told that the mere existence of 0x2a05 will force iOS not to cache that characteristic; unclear if it triggers complete rediscovery.  Perhaps Android has similar behavior, or perhaps you could send indicates from all characteristics to force refresh?

closed 2025-09-16 by weliem