weliem/bluez_inc #98: DESCRIPTOR_METHOD_WRITE_VALUE Multiple calls will crash Because g_variant_unref

ppq19960109, 2025-05-09:
## g_variant_unref release error should be changed to CHARACTERISTIC_METHOD_WRITE_VALUE

## error  code
```
} else if (g_str_equal(method, DESCRIPTOR_METHOD_WRITE_VALUE)) {
        g_assert(g_str_equal(g_variant_get_type_string(params), "(aya{sv})"));
        GVariant *valueVariant, *optionsVariant;

        // Get the options
        g_variant_get(params, "(@ay@a{sv})", &valueVariant, &optionsVariant);
        WriteOptions *options = parse_write_options(optionsVariant);
        g_variant_unref(optionsVariant);

        // Get the byte array to be written
        GByteArray *byteArray = g_variant_get_byte_array(valueVariant);
        g_variant_unref(valueVariant);

        log_debug(TAG, "write descriptor <%s> by %s", localDescriptor->uuid, options->device);

        // Allow application to accept/reject the characteristic value before setting it
        const char *result = NULL;
        if (application->on_desc_write != NULL) {
            result = application->on_desc_write(localDescriptor->application,
                                                options->device,
                                                localDescriptor->service_uuid,
                                                localDescriptor->char_uuid,
                                                localDescriptor->uuid,
                                                byteArray);
        }
        write_options_free(options);

        if (result) {
            g_dbus_method_invocation_return_dbus_error(invocation, result, "write error");
            log_debug(TAG, "write error");
            return;
        }

        binc_descriptor_set_value(application, localDescriptor, byteArray);

        g_dbus_method_invocation_return_value(invocation, g_variant_new("()"));
    }
```

## correct code
```
    } else if (g_str_equal(method, CHARACTERISTIC_METHOD_WRITE_VALUE)) {
        g_assert(g_str_equal(g_variant_get_type_string(params), "(aya{sv})"));
        GVariant *valueVariant, *optionsVariant;

        // Get the write options
        g_variant_get(params, "(@ay@a{sv})", &valueVariant, &optionsVariant);
        WriteOptions *options = parse_write_options(optionsVariant);
        g_variant_unref(optionsVariant);

        // Get the byte array to be written
        GByteArray *byteArray = g_variant_get_byte_array(valueVariant);

        log_debug(TAG, "write <%s>", characteristic->uuid);

        // Allow application to accept/reject the characteristic value before setting it
        const char *result = NULL;
        if (application->on_char_write != NULL) {
            result = application->on_char_write(characteristic->application, options->device,
                                                characteristic->service_uuid,
                                                characteristic->uuid, byteArray);
        }
        write_options_free(options);

        if (result) {
            g_dbus_method_invocation_return_dbus_error(invocation, result, "write error");
            log_debug(TAG, "write error");
            return;
        }

        // TODO deal with offset and mtu
        binc_characteristic_set_value(application, characteristic, byteArray);

        if (byteArray != NULL) {
            g_byte_array_free(byteArray, FALSE);
        }

        g_variant_unref(valueVariant);

        g_dbus_method_invocation_return_value(invocation, g_variant_new("()"));
    }
```

weliem, 2025-05-12:
Could you explain the issue? I don't fully understand...

weliem, 2025-05-18:
Fixed by 821f6d5839f88b708d781330cc0e785ce6ac9f0b

closed 2025-05-18 by weliem