weliem/bluez_inc #99: ERROR [Adapter] failed to register advertisement (error 36: GDBus.Error:org.bluez.Error.Failed: Failed to register advertisement)

shiyu-sz, 2025-05-17:
On some platforms, you need to run the peripheral program twice to broadcast normally.

`[aic@] # example_bluez_inc_peripheral
1970-01-01 08:03:17:613 DEBUG [Adapter] finding adapters
1970-01-01 08:03:17:632 DEBUG [Adapter] found adapter '/org/bluez/hci0'
1970-01-01 08:03:17:633 DEBUG [Adapter] found 1 adapter
1970-01-01 08:03:17:633 DEBUG [Main] using default_adapter '/org/bluez/hci0'
1970-01-01 08:03[   82.797830] rtk_btusb: btusb_open hdev->promisc ==0
:17:637 DEBUG [Application] successfully published application
1970-01-01 08:03:17:637 DEBUG [Application] successfully published local service 00001809-0000-1000-8000-00805f9b34fb
1970-01-01 08:03:17:681 DEBUG [Application] successfully published local characteristic 00002a1c-0000-1000-8000-00805f9b34fb
1970-01-01 08:03:17:688 DEBUG [Application] successfully published local descriptor 00002901-0000-1000-8000-00805f9b34fb
1970-01-01 08:03:17:689 DEBUG [Application] set value <68656c6c6f20746865726500> to <00002901-0000-1000-8000-00805f9b34fb>
1970-01-01 08:03:17:701 DEBUG [Advertisement] setting advertising MinInterval to 500ms (requires experimental if version < v5.77)
1970-01-01 08:03:17:701 DEBUG [Advertisement] setting advertising MaxInterval to 500ms (requires experimental if version < v5.77)
1970-01-01 08:03:17:716 DEBUG [Application] adding /org/bluez/bincapp_hci0_uPlu/service0
1970-01-01 08:03:17:718 DEBUG [Application] adding /org/bluez/bincapp_hci0_uPlu/service0/char0
1970-01-01 08:03:17:727 DEBUG [Application] adding /org/bluez/bincapp_hci0_uPlu/service0/char0/desc0
1970-01-01 08:03:17:777 DEBUG [Adapter] successfully registered application
1970-01-01 08:03:17:789 DEBUG [Main] powered 'on' (/org/bluez/hci0)
1970-01-01 08:03:17:793 ERROR [Adapter] failed to register advertisement (error 36: GDBus.Error:org.bluez.Error.Failed: Failed to register advertisement)`

It can broadcast normally when running it again on ubuntu20, but it can broadcast normally when running it for the first time on ubuntu24. Why is this?

weliem, 2025-05-17:
To be honest, no idea... never seen that behaviour

shiyu-sz, 2025-05-17:
@weliem  I am not the first person to encounter this problem. I upgraded bluez to 5.72 on ubuntu20 and it worked fine. But I still can't do it on buildroot 2022. I still have to run the peripheral program twice to broadcast normally.

https://stackoverflow.com/questions/79532433/bluez-inc-aka-binc-peripheral-sample-dont-work-on-a-raspberry-but-works-on-la

weliem, 2025-05-18:
Should be fixed by ba29bfbc59d069eaeb2caf67cf640b972684207b. Can you try again?

shiyu-sz, 2025-05-18:
@weliem It’s amazing. It runs normally in buildroot2022.04 (bluez5.62). Can you explain why?

weliem, 2025-05-18:
Well, recently the 'SecondaryChannel' property was added to the advertisement object. However, that (apparently) triggers Bluez to use Extended Advertisements rather than 'legacy' advertisements, even of you set 1M. Extended Advertisements are only supported by more recent kernels I think and hence it caused the error. In this commit I only use extended advertisements if secondary channel is set to values other tan 1M. So that fixes it.

closed 2025-05-18 by weliem