weliem/bluez_inc #101: GATT services resolved only once

Vishal8523, 2025-06-05:
Hi,

Environment: BlueZ 5.82 on Debian 12 with Kernel 6.1

I am developing a BLE central application that reads data at regular intervals

I am facing issues in resolving the services and reading characteristics as well. 
Let me explain:
During the 1st connectivity / pair operation, the services are resolved by default

```
2025-06-04 13:22:52:203 DEBUG [Device] Characteristic{uuid='00002a26-0000-1000-8000-00805f9b34fb', flags='[read]', properties=2, service_uuid='0000180a-0000-1000-8000-00805f9b34fb, mtu=156'}
2025-06-04 13:22:52:203 DEBUG [Device] Characteristic{uuid='00002a27-0000-1000-8000-00805f9b34fb', flags='[read]', properties=2, service_uuid='0000180a-0000-1000-8000-00805f9b34fb, mtu=156'}
2025-06-04 13:22:52:204 DEBUG [Device] Characteristic{uuid='00002a25-0000-1000-8000-00805f9b34fb', flags='[read]', properties=2, service_uuid='0000180a-0000-1000-8000-00805f9b34fb, mtu=156'}
2025-06-04 13:22:52:204 DEBUG [Device] Characteristic{uuid='00002a29-0000-1000-8000-00805f9b34fb', flags='[read]', properties=2, service_uuid='0000180a-0000-1000-8000-00805f9b34fb, mtu=156'}
2025-06-04 13:22:52:204 DEBUG [Device] Characteristic{uuid='00002a05-0000-1000-8000-00805f9b34fb', flags='[indicate]', properties=32, service_uuid='00001801-0000-1000-8000-00805f9b34fb, mtu=156'}
2025-06-04 13:22:52:204 DEBUG [Device] Descriptor{uuid='00002902-0000-1000-8000-00805f9b34fb', flags='[]', properties=0, char_uuid='00002a05-0000-1000-8000-00805f9b34fb'}
2025-06-04 13:22:52:204 DEBUG [Device] found 8 services
Services resolved: AA:BB:CC:DD:EE:FF
```

But while reading the data from characteristic, it says:
```
2025-06-04 13:22:52:205 DEBUG [Characteristic] reading <00002a26-0000-1000-8000-00805f9b34fb>
2025-06-04 13:22:52:205 DEBUG [Device] ServicesResolved false
Device: 24:ED:FD:00:06:8B, read operation failed: GDBus.Error:org.bluez.Error.Failed: Not connected
2025-06-04 13:22:52:207 DEBUG [Characteristic] failed to call 'ReadValue' (error 36: GDBus.Error:org.bluez.Error.Failed: Not connected)
```

But the thing is that, the peripheral device is still connected during the time of reading data from characteristic

Also during the 2nd iteration after some amount of time, whenever connect API is called, the services are not at all resolved

What could be the possible root cause of this issue ?

Let me know. Really appreciate your inputs

Vishal8523, 2025-06-06:
@weliem: Any inputs ?

weliem, 2025-06-06:
Looks like your device is indeed disconnecting. Getting 'ServicesResolved false' usually is an indication that the device is disconnecting

What does you code look like?

Vishal8523, 2025-06-09:
After pairing is complete, I disconnect. After a while, I connect and try to read characteristic. So, the subsequent times I connect, the services are not resolved. It is resolved only once during pairing

weliem, 2025-06-09:
Based on what you are saying I cannot tell what is going on. Please share relevant code and a full log.

Vishal8523, 2025-06-11:
Really appreciate your inputs. Yes I have seen the demo examples. I want to actually provide an abstraction interface so that Users can call the functions without worrying about the internal workings. Could you suggest me something ?

closed 2025-06-26 by Vishal8523