weliem/bluez_inc #102: How to Manage/Connect multiple peripherals at the same time ?

Vishal8523, 2025-06-18:
I am trying to develop an application using bluez_inc to connect with 7 devices at the same time in parallel. How to do it ? I see a single main_loop that interfaces with dbus, so how are the requests/errors tracked to each device ? Is it truly parallel or asynchronous ?

weliem, 2025-06-18:
you don't need to do anything special. every callback has a device parameter so that you know from what device the callback is

Vishal8523, 2025-06-20:
Ok. How to associate the error messages from logger to a specific device ?
For example: This is an error log from main loop - `2025-06-09 10:26:07:316 DEBUG [Characteristic] failed to call 'ReadValue' (error 36: GDBus.Error:org.bluez.Error.Failed: Operation failed with ATT error: 0x0e)`
Here there is no mapping of this error to a device. 

How to achieve true parallelism with connect, read_characteristic and disconnect with proper request-response/error mapping ? considering a single instance of g_main_loop running. Any pattern you could suggest ?

weliem, 2025-06-20:
Even if there is an error the callback function for the read operation will be called. There you know the device it concerns...

Vishal8523, 2025-06-24:
@weliem : One more question: I noticed that pairing, connecting have a timeout from bluez_inc library to 24 seconds. How do I reduce it ?

weliem, 2025-06-24:
Not possible, hardcoded inside Bluez

abqjln, 2025-06-25:
Sadly confirm true. I tried variations of Min/MaxConnectionInterval in main.conf to no avail (MaxConnectionInterval = 24 is default).

I had 5 peripheral/server fitness devices connected simultaneously on RPi5 using binc. Extended connection times in an attempt to get this to work on a PiZero. Nope.

Ultimately stock bluez 5.66 values worked—my issues were in banging on dbus thru binc (added enqueing system), and device state management (added cache). I needed another management layer between binc and application code.

Vishal8523, 2025-06-25:
I also see that all dbus calls are not cancellable. 

```
void binc_device_connect(Device *device) {
    g_assert(device != NULL);
    g_assert(device->path != NULL);

    // Don't do anything if we are not disconnected
    if (device->connection_state != BINC_DISCONNECTED) return;

    log_debug(TAG, "Connecting to '%s' (%s) (%s)", device->name, device->address,
              device->paired ? "BINC_BONDED" : "BINC_BOND_NONE");

    binc_device_internal_set_conn_state(device, BINC_CONNECTING, NULL);
    subscribe_prop_changed(device);
    g_dbus_connection_call(device->connection,
                           BLUEZ_DBUS,
                           device->path,
                           INTERFACE_DEVICE,
                           DEVICE_METHOD_CONNECT,
                           NULL,
                           NULL,
                           G_DBUS_CALL_FLAGS_NONE,
                           -1,
                           NULL, // Cancellable to be passed here
                           (GAsyncReadyCallback) binc_internal_device_connect_cb,
                           device);
}
```
There is no way that once I initiate a connection request, it cannot be cancelled. How can I make it happen ? Any suggestions ?

Vishal8523, 2025-06-26:
@weliem : Thanks for all your inputs. I really appreciate it
One question - I see that pairing does not have an error in the callback
For example:
`[3]: Device : failed to call 'Pair' (error 36: GDBus.Error:org.bluez.Error.AuthenticationFailed: Authentication Failed)`
How do I catch such kind of pair related errors ?

closed 2025-06-26 by Vishal8523