weliem/bluez_inc #107: How to get client MTU when my device is a GATT server?

w544503507, 2025-08-22:
Hello,

I have a BLE setup where my device acts as the GATT server, and my phone is the client. When the phone connects to my device, I want to know the MTU of the phone so that I can split data properly when sending.

I tried using the interface binc_device_get_mtu, but it always returns 23.

Is there any way to get the client-side MTU from the server side? I need this value to determine the maximum payload size per ATT packet.

Thanks in advance for any guidance!

w544503507, 2025-08-25:
> Internally I have the MTU per request, I can add it if you want...

That would be great, thanks！An API to get the negotiated MTU on the server side would be really helpful.

Looking forward to it!

weliem, 2025-08-31:
Fixed by 25b720a35daba0331f2bccb8673198df1a76af4e